computourist/RFM69-MQTT-client #1: Gateway stops functioning after a couple of polls/writes

cub69, 2015-01-28:
Hi, nice solution.  I've built both the gateway and the node.  The hardware I'm using is slightly different, I'm using arduino Mega 2560's, a different ethernet device - but the library (UIPethernet) is appartently 100% compatible with ethernet.h.  
The gateway comes up and on my mosquito side I see it begin to send PINGREQ and getting PINGRESP.
I bring up the node and it sends a dev99 and then the temp/humidity.  I will see this publish if the gateway has not already gotten lost.  After a few minutes the gateway just stops - no more ping requests, and no more forwarding of data.  If I do a soft reset it comes back to life until it stops again.  Any suggestions on how to figure this out?  It most often stops after receiving sensor data, but has occasionally stopped without receiving it as well.

Thx

computourist, 2015-01-28:
Hi Cub69,

This is probably a hardware problem. I myself never encountered such problems; my gateway/node hase been running weeks at a time without problems. 
Others have had such ussues, and found solutions. Maybe the discussions on http://homeautomation.proboards.com/ are of use to you...
- Computourist

cub69, 2015-01-28:
thanks found the forum - looking at solutions.

closed 2015-01-28 by cub69