computourist/RFM69-MQTT-client #2: using serial instead of Ethernet

einsiedlerkrebs, 2015-02-21:
Hello Computourist, 

i am happy to find your code, since i was looking for an MQTT(-SN) implementation for arduinos and RFM69. I was wondering how it would be possible, to run the gateway without an Ethernet shield, but an UART-serial instead? The scenario would be, that the GW-Arduino is connected to a third device (i.e. openWRT-machine with broker) using the serial to communicate with the broker.
An advatage is that you could use  a arduino pro mini with 3.3V directly with the radio modules an lower the hardware costs. 

Thank you for the code (!) and for your help.

computourist, 2015-02-21:
Hello Einsiedlerkrebs,

I chose for this setup for the following reasons:
- MQTT is designed to run on top of TCP/IP ; running it over a serial line means a proprietary solution and needs extra code on the receiving end (Mosquitto).
- The gateway is an important part of the installation. I do not like the idea of building a bottleneck (limited serial transfer speed) in such a central part.
- The physical location of the gateway is determined by coverage of the radio network. Your radio gateway will probably not sit right next to your Mosquitto broker, as would be needed for a serial connection. Ethernet gives you the freedom of physically moving the gateway around.
- The extra cost is just one ethernet module. You only need a single gateway per installation, so this would not seem a great problem. 

I know that development is going on to implement a gateway together with a broker on Raspberry Pi. Have a look to the following forum for some inspiration:

http://homeautomation.proboards.com/

Regards,
- Computourist

einsiedlerkrebs, 2015-02-23:
hi, 

i do not mean, to make RFM69-MQTT-client depending on such a setting, but still it would be usefull. In my case i would like to use it in such a way, since

* i have 3.3v Arduinos (Ethernet shield will not work)
* i could port the serial to udp via socat. 

is it not possible, to use such a setting via an ifdef option? I am just asking cause i dont know, ho to change the Ethernet based commands to serial based ones. 

Regards 

einsiedlerkrebs

computourist, 2015-02-23:
Hi,
- you can use the ethershield with a 3.3 Volt Arduino. 
- you cannot change the Ethernet based commands to serial ones. MQTT is based on TCP/IP and you cannot send it over serial. You could build a workaround, but on de broker-machine you would need code to convert the serial back to MQTT. Socat is character based and will not work straight away. 

Rgds,
Computourist

closed 2015-03-03 by computourist