computourist/RFM69-MQTT-client #5: version 2.2 Gateway -- Freezing

davidhrbaty, 2015-04-29:
Hello computourist,
I'm trying your code of Gateway, but I have big strange with freezing after several minutes. When I turned off debug, it was better to take uptime several minutes. Can you advise me, where can be a  problem? I downloaded the newest version RFM libraries, but problem still persist. 

Thank you

computourist, 2015-05-12:
Hi David,

Sorry for the delay in replying; I have been abroad for some time.

It is hard for me to debug your gateway. You would have to indicate exactly
how you put together the hardware and how you implemented your MQTT broker.
Is your gateway faulty or are the end nodes failing ? Is the link with the
broker down or not ?

My experience is these problems in most cases are power supply related.
Make sure your power supply delivers enough current and yout 3.3 Volts
supply for the RFM module can deliver the current needed in high-power
mode. Decoupling of the power supply is needed.

Good luck !

Computourist

davidhrbaty, 2015-05-12:
Hi computourist,

Thank you for your response. I know, that it is hard to debug my gateway. So I include my schematic, because I have used arduino with 5V, so I must use 74HC4050 between arduino board and rfm69hw board. There is schematic:

https://lh4.googleusercontent.com/-vKxMicnYmfQ/VVJO1UJzJ6I/AAAAAAAAEOE/YZsUyeWj1cg/w958-h552-no/rfm69_schematic.png

After main change in your code, especially with mqttclient loop:
```
long watchdogInterval = 2000;

if (millis() > watchdog) {
    watchdog += watchdogInterval;
    if (!mqttClient.connected() && !connect_mqtt()) {
      return;
    }
    noInterrupts();
    mqttClient.loop();
    interrupts();
}
}	// end loop

boolean connect_mqtt() {
   mqttCon = 0;
   digitalWrite(MQCON, LOW);
  if (mqttClient.connect(clientName)) {
      mqttCon = 1;				// Yes, we have a link so,
      digitalWrite(MQCON, mqttCon);	// turn on MQTT link indicator and
      mqttClient.subscribe(subTopic);	// re-subscribe to mqtt topic
    return true;
  }
  return false;
}
```
I got uptime for several days. After four days of operation I got freeze of gateway. My wireless nodes are up, but have no connection. After restart gateway, wireless nodes have connection again. 

When gateway is in freeze mode, it does not respond to any mqtt message. 
MQTT broker (mymousquito) is on the same subnet that wireless gateway.

Thank you for your time.

computourist, 2015-05-13:
Hello David,

have you changed the interrupt handling in file W5100.h before compiling, as indicated in the header of the sketch ?

Rgds,

davidhrbaty, 2015-05-13:
Hi computourist,

Do you mean changes in w5100.h ? Especially this section: 
```
#else
inline static void initSS() { DDRB |= _BV(2); };
inline static void setSS() { cli(); PORTB &= ~_BV(2); };
inline static void resetSS() { PORTB |= _BV(2); sei(); };
#endif
```

Because I forgot to change it. I think, that there will be whole problem. I'll try it when I 'll be at home during weekend. 

I'll write here the result. Thank you again for your advice.

computourist, 2015-05-13:
Without these changes in W5100.h the RFM69 tries to seize the SPI bus at the same time as ethernet.

Gizzbert, 2015-05-30:
ah, thanks, this helped, I also overlooked it, might be helpful to add the following to the sketch of the gateway:
    //                change %program files\arduino\libraries\Ethernet\src\utility\w5100.h:
    //                #else
    //                inline static void initSS() { DDRB |= _BV(2); };
    //                inline static void setSS() { cli(); PORTB &= ~_BV(2); };
    //                inline static void resetSS() { PORTB |= _BV(2); sei(); };
    //                #endif

thanks, gateway works like a charm now :)

computourist, 2015-05-31:
Good to hear you got it all working now. Have fun !

closed 2015-05-31 by computourist