computourist/RFM69-MQTT-client #6: Unable to use both Ethernet Shield and RFM69 - SPI Slave Problem?

TP1337, 2015-05-20:
Hi, I have a problem regarding the radio and ethernet module.
First of all I tried your setup with a Pro Mini (as in the images) and also with an Uno R3 and level converter so that the RFM69 won't get damaged. Both Arduinos are "compatbile" China versions. I tried two China ethernet shields and I also got the genuine ethernet shield.

Which Arduino IDE are you using? Currently I'm using 1.65 and also tried 1.60. I changed two lines in the w5100.h but it won't work. If I unplug the RFM69 power it instantly connects to the MQTT broker, so it's probably due to the SPI problem.

Please tell me which IDE version you are using and if I should try some different ethernet or SPI files. Thank you very much!

computourist, 2015-05-21:
Hi TP,

I'm using an old version of Arduino IDE (1.05), never bothered to upgrade. But that should not cause any problems.

The SPI conflict only occurs when the RFM module receives a radio packet. The RFM will generate an interrupt request and will access the SPI bus. By disabling responses to the interrupt request whilst the ethernet module is using the SPI bus this problem is solved.

This means the connection to the MQTT broker should be established and remain connected until a conflict occurs. If you have no end node transmitting, no packets are received and the connection with the MQTT-broker should not be affected.

I think you have a different problem.
What is the serial output with/without the RFM connected ? Have you checked your supply voltages with/without the RFM ?

Leaving out the LED's is no problem.

TP1337, 2015-05-21:
Hi computourist, 
thank you very much for your quick response! Your comment helps me a lot.

I tried both mb102 as well as usb (and batteries). Maybe it's a good idea to connect it to a better voltage source to be sure it's not a power problem. I will also try IDE 1.05 - all I have to is to set ip, mac etc. and change the two lines in the w5100.h header file, right? Just asking because it doesn't seem to work although I'm trying very hard and I'm experienced with this kind of setups.

If I connect both the rfm69 (not high power version) and ethernet to the pro mini or uno the console only says "failed to connect" once and then kind of freezes. Without the rfm69 it connects immediatedly after one failed connect. I will check everything asap.

Thank you very much for your help!

TP1337, 2015-05-22:
Hi computourist,
I tried several voltage sources but it just wont work. IDE 1.05 didnt help either.

If I use the pro mini I can't even connect to the server - using the same on the UNO sketch I can connect to the mqtt server but only without MISO connected between RFM69 and Arduino UNO. I tried several ethernet shields as well as many boards. Do you think the problem might be related to the boards not beeing genuine ones? The uber homeautomation sketches (Eric Tsai) are working without any problems so far... I really dont know what to do. Do you have any advice for me or could you give me any hints?

I should have the correct wiring between pro mini/ UNO (with level shifter) and I also have set the everything correctly up (mac, mqtt server ip, ...). What else could I try? Its impossible everybody else gets this sketch to work except for me. I also changed the w5100.h according to your description. I even tried HCW and CW versions of the module.

In order to use your setup: can I leave the sketch RFM_MQTT_GW_22.ino as it is except for some values like HCW, mac, ip, password etc.?

Thank you so much!

computourist, 2015-05-25:
Difficult to diagnose your problems. The fact that you get a connection with MISO disconnected suggest a problem with the sharing of the SPI bus. The uber sketches of Eric avoid sharing the bus by using 2 Arduino's connected with a serial link.
Do you use level-shifters ? If so, how are they connected ? Are they 3-state (on,off, float) so you can float the SPI-bus for other devices to use ?
- The gateway sketch can be used as-is; only thing you need to change are some parameters.
- The MQTT-connection always shouts: "no connection" before connecting.. (bad programming) 
- The boards I use are China-clones; no problem with that...

Success....

closed 2015-06-16 by computourist