computourist/RFM69-MQTT-client #8: Arduino IDE 1.6.7 compiler failure

skywayne, 2016-01-23:
Trying to compile the Gateway with Arduino IDDE 1.6.7 gives the error 'mqtt_subs' not declared in this scope for the line -PubSubClient mqttClient(mqtt_server, 1883, mqtt_subs, ethClient );

Moving the function above this line then gives an error as mqttclient is not declared and is used in mqtt_subs.  I seem to be chasing my tail here?  Any ideas?

militis01, 2016-01-23:
I ran into the same problem. I solved it by defining a prototype of the Mqtt_subs function above the mqtt_client instantiation.

skywayne, 2016-01-23:
Thanks - I will try that and see if Gateway fully operational.

computourist, 2016-01-23:
Seems the arduino IDE is handling things a bit different now.
Have a look at:
http://homeautomation.proboards.com/thread/94/solved-mqtt-subs-declared-scope

Computourist

computourist, 2016-01-31:
Released version 2.4 where this problem is solved.

closed 2016-01-31 by computourist