computourist/RFM69-MQTT-client #10: Gateway up or down

bcomsugi, 2016-03-03:
Hi, how do you know if the gateway is up or down?
i can't ping the gateway, and the gateway seem didnt implement reconnect if the connection(LAN) is disconnected

militis01, 2016-03-04:
I am using an Mqtt client (lens) in chrome browser to send status request messages to the gateway. The documentation on the gateway covers how to do that.

computourist, 2016-03-04:
You should be able to ping the LAN-connection of your gateway. If this is not the case, do you have a link-indicator on your ethernet port ? Is you MAC-adrress (as defined in the sketch) unique in your network?
If the ethernet port is up, the MQTT LED will indicate connection to you broker. In debug mode, a connection message will be present on the serial port.

closed 2016-08-04 by computourist