computourist/RFM69-MQTT-client #12: Arduino compiler error

dome2048, 2017-02-07:
I'm using:

Mac OS X 10.12.3
Arduino IDE 1.8.1
Arduino Uno (using logic level converters and with radio connected to 3.3v)

I installed the RFM69 library manually and the SPI library through the library manager. When I go to compile RFM_MQTT_GW_25, I get the following error log:

`
/Users/dome/Documents/Arduino/libraries/RFM69-master/RFM69_OTA.cpp: In function 'void HandleHandshakeACK(RFM69, SPIFlash, uint8_t)':
/Users/dome/Documents/Arduino/libraries/RFM69-master/RFM69_OTA.cpp:80:16: error: 'class SPIFlash' has no member named 'initialize'
     if (!flash.initialize())
                ^
/Users/dome/Documents/Arduino/libraries/RFM69-master/RFM69_OTA.cpp: In function 'uint8_t HandleWirelessHEXData(RFM69, uint8_t, SPIFlash, uint8_t, uint8_t)':
/Users/dome/Documents/Arduino/libraries/RFM69-master/RFM69_OTA.cpp:124:9: error: 'class SPIFlash' has no member named 'blockErase32K'
   flash.blockErase32K(0);
         ^
/Users/dome/Documents/Arduino/libraries/RFM69-master/RFM69_OTA.cpp:125:9: error: 'class SPIFlash' has no member named 'writeBytes'
   flash.writeBytes(0,"FLXIMG:", 7);
         ^
/Users/dome/Documents/Arduino/libraries/RFM69-master/RFM69_OTA.cpp:174:50: error: 'class SPIFlash' has no member named 'blockErase32K'
                 if (bytesFlashed%32768==0) flash.blockErase32K(bytesFlashed);//erase subsequent 32K blocks (possible in case of atmega1284p)
                                                  ^
exit status 1
Error compiling for board Arduino/Genuino Uno.`

I know there's been a lot of changes with the Arduino IDE lately; any idea why this broke or if it's simply user error?

computourist, 2017-02-08:
Did you load the correct libraries ? Somehow you are pointing to Flash memory on the SPI bus ??
I have never seen such references..

dome2048, 2017-02-09:
Okay, so I did get it to compile. I started with a fresh install of the Arduino IDE on a Raspberry Pi, and manually installed the RFM69.h, SPI.h, PubSubClient.h, and SPIFlash.h libraries from their respective Github sources.

I had to edit the library names and file locations for the IDE to see them, and had to add `#include <SPIFlash.h>` in order for the code to compile.

I do have quite a few libraries installed on my Mac Arduino IDE so it's possible there's some type of conflict.

Based on your advice from the other open issue, I'm waiting for my 3.3v Pro Micro and 3.3v w5100 module before I proceed to hardware testing, rather than fiddle with logic level converters.

If you think it would be helpful (perhaps I'm not the only one with this issue?), I'd be happy to do a PR for the readme file to explain how I got it to compile.

waynestronach, 2017-02-09:
I have built about 20 or so Nodes and gateways using RFM69HW, RFM69HC, and nanos, unos, megas of both original and clones (e.g sparfun) types.

I initially brought level converters, but haven't installed a single one and have had no problems over the last 24 months.  A 5V Arduino needs a voltage > 3V to register a 1(true).  3.3V with open collector (as per the RFM69s) works OK.

One issue to be aware of with the gateway is the need to edit the Arduino Ethernet library to ensure that the interrupts work correctly - see notes on the gateway or http://harizanov.com/2012/04/rfm12b-and-arduino-ethernet-with-wiznet5100-chip/

consolacion, 2017-02-13:
Getting the same 'class SPIFlash' has no member named .........' errors as dome2048.  Used the SPIFlash library from the library manager. Should i use another??

@dome2048 yes how did you solve it??

dome2048, 2017-02-16:
@consolacion I got it to compile on my mac by removing all my libraries and using a temporary sketch folder with only the 5 libraries, and with versions downloaded from GitHub, not the library manager. Sounds like we both found our way to the same solution though.

It does bear mentioning that I still needed to add `#include <SPIFlash.h>` to get the sketch to compile. I'll submit a PR for this.

consolacion, 2017-02-16:
Thanks, and indeed, I had found proper library already :-) However, I did not have to include the SPIFlash to the sketch.
So many libraries with same name :-)  beware that 'updating' through the library manager sometimes updates the wrong library

dome2048, 2017-02-16:
@consolacion Ah, I see - I reread your post and it makes more sense.

I'll close this issue. Thanks for the help everyone.

closed 2017-02-16 by dome2048