computourist/RFM69-MQTT-client #13: Add schematic for Uno (5v) with level shifters + RFM

bpridemore, 2017-02-08:
Arduino 's running 5v e.g. Uno are popular, please add a drawing to the code base for wire hook ups thais including level shifters with RFM

Gizzbert, 2017-02-08:
RFM69HW does not work with the regular 5v uno due to higher input current
requirements,  it needs 150mA.

computourist, 2017-02-08:
RFM69 is also likely to be damaged when IO ports are hooked up to 5 volts.
I have never used a setup with level shifters, so I cannot provide a schematic for a 'working' setup.
It might be easier to convert your UNO to 3.3 Volts. 
Have a look at: https://learn.adafruit.com/arduino-tips-tricks-and-techniques/3-3v-conversion

bpridemore, 2017-02-08:
https://forum.arduino.cc/index.php?topic=373128.0
I'm pretty sure it's possible with level  shifters or converters.

computourist, 2017-02-08:
Yeah, I'm sure too.
In the gateway three devices (Uno, ethernet and RFM69) are connected to the same SPI signals. There is a need to 'float' the bus if no communication is taking place. This means extra restrictions on the way level shifters are used. Believe me, using a 3.3 Volt uno is a far better way to go.

computourist, 2017-02-08:
You might want to have a look at:
http://homeautomation.proboards.com/thread/114/logic-level-converters-rfm69hw
where a working config is dicussed.

consolacion, 2017-02-13:
for Node I use an 8MHz 3v3 Pro Mini, but as i want to use my arduino ethernetshield, I am kinda stuck on using a Uno and was going to use a levelshifter BOB or a CD4050. But as I like to keep my wiring simple, it might be a better idea to get a W5100 module and a 3v3 pro mini.
Having said that.. knowing the W5100 module will cost 5-8 euro, maybe an esp8266 Gateway might be a better idea..... hmmm got some thinking to do

closed 2017-03-31 by computourist

computourist, 2017-04-12:
Hi Consolacion,

My feedback form the forum is that many find implementing the ESP8266 node a simpler way to start. The ESP8266 node incorporates an MQTT client and therefor needs no gateway. In this way a lot of  complexity is avoided.

consolacion, 2017-04-12:
Tnx yes indeed. Initially I came upon many people  experiencing problems with it, but I just now built another gateway with an Wemos D1 and that gave no problems.
By the way, I haven't been buying Elektor for a long time, but I may coz of yr article