computourist/RFM69-MQTT-client #14: SIM800 GSM module

esawyja, 2017-03-30:
Would it be possible to send the sensor values to the Cloud directly via a SIM800 GSM module? IE for use where there is no internet available, if possible in MQTT format?

computourist, 2017-03-31:
Sure, it would be possible. But since MQTT needs a permanent connection, it would deplete your dataplan very soon. 
A better way would be to collect and store your data locally and connect on a regular basis to upload your data using a dedicated api. In this way your connection cost over the GSM network can be limited.

esawyja, 2017-03-31:
Thanks, I did not know MQTT needs a permanent connection, I thought it will just be a short burst of info when you publish?

computourist, 2017-03-31:
If there is no permanent connection, the broker can never push a message. The client initiates the connection to the broker...

esawyja, 2017-03-31:
aaah ok, what I'm looking for is just the client part to work on a GSM
modem, so I would have the broker running in the Cloud and the Arduino and
SIM800 as a client, pushing sensor data to the broker

computourist, 2017-03-31:
I guess you could initiate an MQTT connection from your client, then tear down the connection after publishing your data. Not sure how Mosquitto would handle that and it seems overkill for just a data upload. Better to place the MQTT client in the cloud with a permanent connection to the broker and upload your data via API.

esawyja, 2017-03-31:
Ok, thanks, I will try that