computourist/RFM69-MQTT-client #15: Elektor publishing your project / WIZ5100 bug /433Mhz node

consolacion, 2017-04-12:
Magazine Elektor in its dutch March/April 2017 is publishing your 'system'.
When they talk about 'the author', I presume it is you. (if not you better have a talk with then)
https://www.elektormagazine.nl/magazine/elektor-201703/40186.
Interesting though through the article I came across the "[WIZ5100 bug](http://john.crouchley.com/blog/archives/662)".  Fortunately I have not been plagued by that, but anybody  that has trouble getting things to work may want to have a look at John Crouchley's blog on the matter.

Elektor does mention an 433MHz Rf endnode. Any chance of you publishing that here too?

computourist, 2017-04-12:
Your assumption is right: Elektor asked me to describe the system in an article.
The WIZ5100 is a sneaky one. At first my gateway seemed to work OK, but every once in a while (days to weeks) it would hang for no obvious reason. After implementing the proposed solution my gateway works witghout problems for months at a row.
The 433 MHz Rf end node is described in: https://github.com/computourist/RFM69-MQTT-client/tree/master/Remote%20Control%20node.
I use it to control several KlikAanKlikUit wall switches I have lying around from earlier projects.

Regards,
Computourist

consolacion, 2017-04-12:
It is a very nice article. Congrats.
My experience with the WIZ5100 is similar, worked well, but when I just tried another module it didnt. The difference was a 74LVC14 (low voltage inverter SchmittTrigger). I have that breadboarded now and problem is gone.
I must have missed the 433Mhz node. tnx

closed 2020-06-24 by computourist